vroland/MyoBridge #2: Have issue please help

atfresh279, 2017-02-04:
Hello, I have been working on the myo bridge and have encountered some issues. I have plugged everything in right, and flashed the firmware to my hm-11 correctly. When I then upload Print firmware and open serial monitor, it only says searching for myo and doesn't do anything else. My myo is connected to the HM-11 board because it has the solid blue rectangle light at the bottom  of the myo and it's isn't not connected to anything else. Any help would be great, please email me at atfresh279@hotmail.com
Thanks

vroland, 2017-02-06:
Are you sure your serial connection is set up correctly? Do you use a level converter or a voltage divider? If you connect a LED to TX and Ground (in line with a resistor), does it flash?

atfresh279, 2017-02-09:
Sorry for the late reply, I ordered another hm-11 board and replace the old one. It instantly worked, it must have had a problem.
Thanks

vroland, 2017-02-09:
Good to hear that!

closed 2017-02-09 by vroland