vroland/MyoBridge #3: General Arduino BLE Central Library

WALLTECH, 2017-03-31:
Hello!
I've been on a quest to use my arduino as BLE central, master device in order to communicate with other peripherals by sending say, notification enable,to a specific characteristic of like a BLE sensor and your genious Myo library is the closest anyone has ever come to it. Do you know of a general way to use the hm-11 and arduino to do what I'm looking for generically? Would it me too much trouble if you cut down this library and firmware perhaps so you can perform its basic functions? This would be most helpful, and a first for the arduino community. Thanks!

vroland, 2017-04-02:
Hallo,

I think the best way to do this would be to write your own firmware for the 
HM-11, as I did. You can use my Library as a basis. Unfortunately, you need 
the IAR Workbench (or at least it's compiler), which is only available as a 30 
day trial if you don't want to pay a horrendous amount of money. This is what 
I did, thus I don't have access to the necessary tools any more. But feel free 
to try it on your own ;)

Am Freitag, 31. März 2017, 04:03:08 CEST schrieb John Wall:
> Hello!
> I've been on a quest to use my arduino as BLE central, master device in
> order to communicate with other peripherals by sending say, notification
> enable,to a specific characteristic of like a [...]

WALLTECH, 2017-04-02:
Thanks for responding. I ended up just buying an Arduino 101 and it worked flawlessly right away with the central library.

closed 2017-04-04 by vroland