vroland/MyoBridge #5: Turn Myo off

martingorosito, 2017-06-08:
Hi, is there any command to turn myo off via arduino?
Regards!

vroland, 2017-06-09:
As far as I know (According to the official Bluetooth protocol), myo only exposes locking/ unlocking and enable/ disable sleep. So no, except if you find a bug in their firmware to crash it you can't ;)

martingorosito, 2017-06-12:
I saw this in the myohw.h bluetooth protocol, I was wondering if you could help me on how to write it for arduino (I am not very strong in programing)

```
typedef enum {
    myohw_command_set_mode               = 0x01, ///< Set EMG and IMU modes. See myohw_command_set_mode_t.
    myohw_command_vibrate                = 0x03, ///< Vibrate. See myohw_command_vibrate_t.
    myohw_command_deep_sleep             = 0x04, ///< Put Myo into deep sleep. See myohw_command_deep_sleep_t.
    myohw_command_vibrate2               = 0x07, ///< Extended vibrate. See myohw_command_vibrate2_t.
    myohw_command_set_sleep_mode         = 0x09, ///< Set sleep mode. See myohw_command_set_sleep_mode_t.
    myohw_command_unlock                 = 0x0a, ///< Unlock Myo. See myohw_command_unlock_t.
    myohw_command_user_action            = 0x0b, ///< Notify user that an action has been recognized / confirmed.
                                                 ///< See myohw_command_user_action_t.
} myohw_command_t;
```

```

typedef struct MYOHW_PACKED {
    myohw_command_header_t header; ///< command == myohw_command_deep_sleep. payload_size == 0.
} myohw_command_deep_sleep_t;
MYOHW_STATIC_ASSERT_SIZED(myohw_command_deep_sleep_t, 2);
```

vroland, 2017-06-15:
Look at the sources of the MyoBridge Arduino library. This is exactly the header I am using. But except for deep sleep, you can use every one of these commands from the High-Level Arduino library.

closed 2017-07-28 by vroland