vroland/MyoBridge #7: Misunderstanding about  the connection between the  HM-11 et Myo

Jo31000, 2017-07-26:
Goodmorning Vroland,

I work on a project of a prosthetic hand and  I try to have a direct connection between my MYO and  my HM-11. I follow all your instruction about how to connect and to flash my  HM-11 thanks to the CC Debugger but at  the end of yours instructions I  don't understand how to connect these two components without my laptop. I need this connection because I want to remove the fact to use my laptop for my prosthetic hand. 
My  HM-11 is called BASIC#7C on the Bluetooth device but when I started Arduino it doesn't recognize my HM-11. So I come to  you to know if you have some information to help me to  do  the direct connection between HM-11 and MYO  after I  had flashing my HM-11.

I wish you a good  day  
Best Regards

nullp0tr, 2017-07-28:
https://blog.raquenaengineering.com/getting-started-with-the-myo-gesture-control-armband/
You'll find tutorials here regarding that, hope it helps you!

nullp0tr, 2017-07-28:
I'd also close the issue to be honest since this is rather a stackoverflow kind of question and not really an issue with the firmware nor the library.

vroland, 2017-07-29:
This looks like a very detailed tutorial indeed, well done!

closed 2017-07-29 by vroland