vroland/MyoBridge #8: Sample rate EMG values

fgaetani, 2017-12-26:
Hi Vroland,
I think that I found an issue in your MyoBridge Library. I'm doing several tests by streaming EMG through Myo Armband, and in these test I should obtain 200Hz sample rate, but I have measured (with several board - Arduino Uno, Mega and Micro) 370Hz for the firsts 4 values and 10Hz for the fifth value. Hence, the sample rate is not regular, and is not of 200Hz, and that hole of 10Hz is a big issue for my application. I tried to find the problem in the library, but I did not succeed. I also noticed the same problem with IMU streaming, but with different frequencies. Do you know where is the problem? Can you help me please?
Thank you very much!

vroland, 2017-12-31:
Hi fgaetani,
I'm afraid I can't really help you with that. It could be some input or output buffering of one of the serial connections, but since I do not own a Myo wristband I cannot test it. Finding and fixing a bug without beeing able to test anything is kind of hard ;)
But in case you find anything, please let me know and / or file a pull request!

fgaetani, 2018-01-13:
Thank you very much, I will try to solve the problem.

closed 2018-01-13 by vroland

fgaetani, 2018-01-18:
I think that I have found the problem. This one is in Myobridge firmware of CC2541 SoC. How can I recompile the firmware with the changes to test it?

vroland, 2018-01-18:
You need the IAR C compiler.. I think you can get a 30 day trial version for free, which I used to develop the original firmware.

ToWey11, 2018-06-11:
Hi fgaetani,
i have similar problems. I only get 50 hz emg data and about 40 hz imu data. Curiously the imu data even starts first with 25 hz and increases continuously up to 40hz.

Did you solve your problem and could explain here how you did it?

I would be very grateful.

Domadrona, 2018-09-10:
hi @fgaetani 
i have the same problem. did you make it work on a higher frequency than 100hz with constant sample rate?. @vroland can you please give to me a guide in how to solve this problem?

thanks you all.

vroland, 2018-09-11:
As I said before, I cannot currently provide support for MyoBridge, since I neither own a Myo Band nor the IAR compiler. :/

Domadrona, 2018-09-11:
@vroland thanks for the quick answer, i do have the armband and i can take the 30 days trial for the compiler. so, what im asking for is: can you offer to me an advise about Wich steps to follow in order to do a good debugging?. i mean, what probably can be the problem. and wich part of the firmware should i focus in?.My team thanks you too much and we really appreciate your help.
best regards

vroland, 2018-09-12:
The example code for the CC2541 provided by TI was really helpful to me. Maybe collect some statistics about the amount of data coming from the Bluetooth connection and leaving through the serial port and see if they differ?

kwalker96, 2019-01-24:
@fgaetani did you have to change anything to connect with the arduino mega? I am able to establish connection with the uno and extract data but when using the mega the data stream does not show in the serial monitor