vroland/MyoBridge #9: Get the MyoBridge Library to work on stm32duino IDE

NBProjects, 2018-03-16:
Hello vroland,

For a university project, I'm trying to get your MyoBridge Library up and running on an STM32 Nucleo F303RE Board by using the stm32duino project (https://github.com/stm32duino/BoardManagerFiles).
First, I programmed an Arduino UNO with the MyoBridge Library and it works fine so far. Then I installed the stm32duino package to the Arduino IDE, tried some basic programs and it also works fine.
But when I want to compile the MyoBridge (basic) projects with the stm32duino board manager, then I get several compiling errors because of missing libraries like "SoftwareSerial.h", "util/delay_basic.h" or "avr/interrupt.h".
I included these libraries manually but there comes up a new error message every time when I try to compile. Can you help me and us by get this running please?

MfG
Noah Boroske

vroland, 2018-03-19:
Hey NBProjects,
As fpistm already mentioned in the BoardManager issue, you probably have to exchange some AVR-specific libs for stm32-based ones. The SoftwareSerial library, for example, is only used because most AVR-based Arduinos do not have a second UART which would be needed for bridge operation. You can use any hardware serial port instead. 
I don't know about the rest, since I can't test anything myself. But I would imagine you can find replacements for the other libraries as well.
I hope this helps, closing this for now.
Greetings,
Valentin

closed 2018-03-19 by vroland

NBProjects, 2018-03-29:
Many thanks for your answer! I will try to replace the SoftwareSerial library, for example with the HardwareSerial library. Hopefully this let compile the MyoBridge sketch.

Regards
Noah

vroland, 2018-05-23:
Looks like there is a name collision with some DEFINEs. You could try to fork MyoBridge and rename every occurrence of SUCCESS to something else...

NBProjects, 2018-06-05:
Thank you for your answer!