vroland/MyoBridge #11: Trying to connect Myo with ESP32 Arduino

Harinka, 2018-05-30:
Hi,

I'm trying to connect Myo with an ESP32 Arduino to read EMG data.

I tryed to use examples from the library, but they uses a serial port : 
`#include <MyoBridge.h>
#include <SoftwareSerial.h>

#define rxPin 2
#define txPin 3

//SoftwareSerial connexion à MyoBridge
SoftwareSerial bridgeSerial(rxPin, txPin);

//Initialise un objet MyoBridge
MyoBridge bridge(bridgeSerial);`

But ESP32 is not compatible with SoftwareSerial due to the BLE included on it.
I can't create a MyoBridge object without a SerialPort ?
Do you know how can I connect Myo with ESP32 ?

Sorry for my bad English 🥇

vroland, 2018-05-30:
Hi Harinka,

Do you have any kind of serial UART communication available? Hardware and simulated (Softare-) ports will do. Maybe there is a similar library for ESP32?
If not, you cannot use this library as is.
Regards,
Valentin

Harinka, 2018-05-30:
Indeed, there is a specific library for ESP32, but I haven't been able to make it works with MyoBridge.

Maybe you can help me : [http://www.instructables.com/id/ESP32-BLE-Android-App-Arduino-IDE-AWESOME/](url)

I tried to follow this tutorial (I just skipped the app part) and change the program to make it work with MyoBridge. But it didn't work

closed 2018-05-30 by vroland