vroland/MyoBridge #14: My Myo doesn´t pair

Sld1c, 2018-09-13:
My Myo device doesn´t pair, I have the Module that is provided by DSD TECH and I already Upload de Firmware to it, but when I try to use the "printFirmwareInfo" example my Myo just doesn´t pair
This is de Module that I have 
[image]
And this happen in my serial monitor 
[image]

Sld1c, 2018-09-13:
I will be very greatful if somebody can help me

vroland, 2018-09-13:
As I do not own a Myo, I'm afraid I can't help you a lot with this. But there are a couple points you could check:
- Are you using the correct chip? This libary requires the CC2541 SoC.
- Is it properly flashed?
- Is the serial communication set up correctly?
- Is everything soldered well?

closed 2018-09-13 by vroland