vroland/MyoBridge #16: How can I get filtered EMG data in order to create new gestures?

Sld1c, 2018-10-11:
I have been trying to create new gestures but I don't have idea how to get an average from the EMG data when I do a diferen pose whit mi hand and using the getEMGdata example in order to create and recognize a new gesture.
If you can help me I will be very gratefuly

vroland, 2018-10-12:
I implemented something like this on the basis of MyoBridge over here: https://github.com/vroland/MyoIMUGestureController.
Hope that helps :)

Sld1c, 2018-10-13:
Thanks a lot, let me che it, I´ll keep you update

umaires28, 2018-10-31:
sir i am working on this can we use the HM-10, Bluetooth module using TI CC2540 chip  instead of HM-11 for connecting myo directly to arduino without any bridge(laptop).
and please guide me  will this all procedure work fine and connect the myo arm band directly to arduino
thanks sir