vroland/MyoBridge #17: can we use any other Bluetooth module for myo bridge

umaires28, 2018-10-31:
sir i am working on this can we use the HM-10, Bluetooth module using TI CC2540 chip instead of HM-11 for connecting myo directly to arduino without any bridge(laptop).
and please guide me will this all procedure work fine and connect the myo arm band directly to arduino
thanks si

vroland, 2018-11-01:
I have not tried using the HM-10 myself, but someone on the thalmic forums may have tried it... But as I do not have access to the device  nor the compiler any more, I'm afraid I cannot help you with building a new firmware. But it might be as easy as downloading the IAR compiler trial and compiling for the CC2540.

kasi09, 2018-11-01:
You can try this ->
https://github.com/kasi09/Myo_ESP32
but I haven't document it until now.
It should work with an ESP32 and when you edit the 
"BLEAddress pAddress = BLEAddress("f9:64:aa:5e:d8:ef");" in "myo.h"

closed 2018-11-02 by vroland

umaires28, 2018-11-02:
sir if i use HM 11 then will i able to connect myo to arduino?
i am making my final year project on it and making a robotic arm for disable persons.

umaires28, 2018-11-02:
> I have not tried using the HM-10 myself, but someone on the thalmic forums may have tried it... But as I do not have access to the device nor the compiler any more, I'm afraid I cannot help you with [...]

sir if i use HM 11 then will i able to connect myo to arduino?i am making my final year project on it and making a robotic arm for disable persons
--

vroland, 2018-11-03:
No need to use a bigger font size here... 
The library should work on a bluetooth module using the cc2541 SoC and was tested with the HM-11. So you should be able to use this library to extract EMG and IMU data from the Myo, 
But I cannot give you any guarantees.This library was developed 3 years ago, with Myo firmware from 3 years ago.

umaires28, 2018-11-03:
> No need to use a bigger font size here...
> The library should work on a bluetooth module using the cc2541 SoC and was tested with the HM-11. So you should be able to use this library to extract EMG [...]

thanks sir