vroland/MyoBridge #18: Cannot connect to Myo

kwalker96, 2019-01-23:
Hi,

I have loaded the MyoBridge firmware to the chip and have then connected the arduino to the chip as described, but my serial monitor reminds in the "Searching for Myo..." stage.

I have tried Davide's library (I am using Mac) but still no luck - any ideas on what may be wrong? I am trying to connect the myo directly to arduino to prototype a low cost prosthetic hand for my final year project.

Thanks

vroland, 2019-01-24:
Debugging remotely is not easy ;) I would suggest trying:
* Can you monitor the serial link between HM-11 and Arduino? Are there any messages sent?
* Are you using the correct chip and firmware?
* Are you sure myo is not connecting to something else?

kwalker96, 2019-01-24:
So we have established a connection between the myo and the uno, but my question is do you know if you can you use a mega instead?

We are trying to read data through a mega as we intend to use the same MCU on our project but having no luck; do you have any tips on anything that may need altered?

vroland, 2019-01-25:
MyoBridge uses a simple serial connection, which is emulated using SoftwareSerial on the Uno. You could try using a hardware serial port if the Mega has one or check the pins you are using support SoftwareSerial.

olivierallard, 2019-01-31:
Hi, ive been trying to make the Arduino communicate with the HM-10 to use with myo armband. I followed the instruction and flashed it, all connections have been checked twice. I get stuck in the searching for myo screen…. Does it only work with HM-11?
How can I monitor the serial link beetween the Arduino and the HM-10?

vroland, 2019-02-01:
As far as I know, the HM-10 uses the CC2540 instead of the 2541 SoC. So the precompiled firmware will not work for these chips. You can try to build the firmware yourself (IAR compiler). If you are lucky, someone has a 2540 firmware around, but that's not me.

simbolonmartin, 2019-03-04:
I am facing the same problem that my Serial Monitor stuck in "Searching for Myo..."
How can I monitor the serial link between the Arduino and the HM-11?
I am using Arduino Uno, and RX Led is not blinking/ on
thanks in advance

simbolonmartin, 2019-03-06:
> I am facing the same problem that my Serial Monitor stuck in "Searching for Myo..."
> How can I monitor the serial link between the Arduino and the HM-11?
> I am using Arduino Uno, and RX Led is not blinking/ on
> thanks in advance

I already fixed the problem, the problem is wrong RX/TX pin due to SerialSoftware library.
If anyone is facing the same problem perhaps this will help

simbolonmartin, 2019-03-06:
> Hi, ive been trying to make the Arduino communicate with the HM-10 to use with myo armband. I followed the instruction and flashed it, all connections have been checked twice. I get stuck in the [...]

i actually using HM-10 Module and work just fine, have you make sure that the RX/TX pin that you are using is the SoftwareSerial pin?

Melunis, 2019-03-23:
can someone help me i am getting the following message in my serial monitor:
Searching for Myo...
CONN_STATUS_DISCOVERING
CONN_STATUS_BRIDGE_SETUP
CONN_STATUS_SCANNING
CONN_STATUS_INIT
CONN_STATUS_SCANNING

what could be the problem?

simbolonmartin, 2019-04-08:
> So we have established a connection between the myo and the uno, but my question is do you know if you can you use a mega instead?
> 
> We are trying to read data through a mega as we intend to use the same MCU on our project but having no luck; do you have any tips on anything that may need altered?

Did you make the RX TX pin in  2 and 3?
Because Arduino Mega cannnot use this pins in SoftwareSerial

alexis-97, 2020-03-23:
Hi. I am trying to do it with an HC-06. I have done the same that you but my Serial Monitor stuck in "Searching for Myo...". Could someone help to me? I don't know what I have to change in my script because of use the HC-06

vroland, 2020-04-01:
The HC-06 won't work. As it is based on a different chip, you will not be able to flash the firmware. Furthermore, Myo is based on Bluetooth LE, while the HC-06 only supports Bluetooth 2.0.

alexis-97, 2020-04-01:
Ok. I suppose Hc-05 won't work neither, right?

shoaibmirza, 2023-02-02:
> I already fixed the problem, the problem is wrong RX/TX pin due to SerialSoftware library. If anyone is facing the same problem perhaps this will help

In my case, I tried changing the software serial pins, **but it did not work**.  Also, my MYO band is not pairing with the HM-10 Bluetooth (the charging LED is off). but when I am connecting the Bluetooth dongle to my PC, the LED turns on (blue-ish color). so I'm guessing it's not connecting with my HM-10 Bluetooth.

Do I have to update the MYO band firmware as well? or something else?

Thank you in advance!

I am using the HM-10 with an Arduino UNO and the MYO Armband (MYOD5).

moaz-hamdy, 2023-12-02:
> i actually using HM-10 Module and work just fine, have you make sure that the RX/TX pin that you are using is the SoftwareSerial pin?

do I have to use the Bluetooth dongle that comes with the MYO to successful connect the to the Arduino ?

moaz-hamdy, 2023-12-02:
did anyone used the MYO to control a prosthetic arm with an Arduino I could use some help :) ?
think you.