vroland/MyoBridge #21: How Can i get both EMG and IMU data at the same time?

simbolonmartin, 2019-03-20:
I see your example code from the library EMG and IMU data is working independently. Can i get both EMG and IMU data at the same time with this library?
Thank you

vroland, 2019-03-20:
You should be able to... Have you tried it?

simbolonmartin, 2019-03-20:
Thanks for responding Sir 🙇 

I tried with just adding the function that handles IMU data to readEMGData.ino in void setup() :

```ino
  //set the function that handles EMG data
  bridge.setEMGDataCallBack(handleEMGData);
  //tell the Myo we want the filtered EMG data
  bridge.setEMGMode(EMG_MODE_SEND);

  //set the function that handles the IMU data
  bridge.setIMUDataCallBack(handleIMUData);
  //tell the Myo we just want the IMU data
  bridge.setIMUMode(IMU_MODE_SEND_DATA);
```

the whole .ino code is:
```
**
 * @file   readEMGData.ino
 * @author Valentin Roland (webmaster at vroland.de)
 * @date   September-October 2015
 * @brief  Example file demonstrating the EMG data streaming feature.
 *
 * This file is part of the MyoBridge Project. For more information, visit https://github.com/vroland/MyoBridge.
 *
 * @include readEMGData.ino
 */
 
#include <MyoBridge.h>
#include <SoftwareSerial.h>

//SoftwareSerial connection to MyoBridge
SoftwareSerial bridgeSerial(2,3);

//initialize MyoBridge object with software serial connection
MyoBridge bridge(bridgeSerial);

//declare a function to handle EMG data
void handleEMGData(int8_t data[8]) {
  
  //the EMG data is 8bit signed integers. Just print them out:
  Serial.print("EMG:  ");
  Serial.print(data[0]);
  Serial.print(" ");
  Serial.print(data[1]);
  Serial.print(" ");
  Serial.print(data[2]);
  Serial.print(" ");
  Serial.print(data[3]);
  Serial.print(" ");
  Serial.print(data[4]);
  Serial.print(" ");
  Serial.print(data[5]);
  Serial.print(" ");
  Serial.print(data[6]);
  Serial.print(" ");
  Serial.print(data[7]);
  Serial.println(" ");
}

void handleIMUData(MyoIMUData& data) {
  
  //Accelerometer data is scaled from 0 to 2048. But just print it out for the Moment:
  Serial.print("IMU:");
  Serial.print(data.accelerometer[0]);
  Serial.print(" ");
  Serial.print(data.accelerometer[1]);
  Serial.print(" ");
  Serial.println(data.accelerometer[2]);
}

void setup() {
  //initialize both serial connections
  Serial.begin(115200);
  bridgeSerial.begin(115200);

  //wait until MyoBridge has found Myo and is connected. Make sure Myo is not connected to anything else and not in standby!
  Serial.println("Searching for Myo...");
  bridge.begin();
  Serial.println("connected!");

  //set the function that handles EMG data
  bridge.setEMGDataCallBack(handleEMGData);
  //tell the Myo we want the filtered EMG data
  bridge.setEMGMode(EMG_MODE_SEND);

  //set the function that handles the IMU data
  bridge.setIMUDataCallBack(handleIMUData);
  //tell the Myo we just want the IMU data
  bridge.setIMUMode(IMU_MODE_SEND_DATA);

  

  
  //disable sleep mode, so we get continous data even when not synced 
  bridge.disableSleep();
}

void loop() {
  //update the connection to MyoBridge
  bridge.update();
}
```

and the result in Serial Monitor it prints EMG Data twice first, and then IMU Data twice,  like this:

EMG:  -14 -2 -2 -11 -4 0 -2 -3 
EMG:  0 1 0 -2 -2 -3 -2 -1 
IMU:-402 -811 1878
IMU:-404 -807 1887
EMG:  -5 -7 -8 0 0 0 -2 -2 
EMG:  13 4 1 -3 -1 -1 -1 0 
IMU:-403 -809 1888
IMU:-397 -801 1874
EMG:  -1 1 1 7 -3 -3 0 -1 
EMG:  -11 -7 5 2 -5 0 -1 -3 
IMU:-408 -813 1883
IMU:-400 -801 1873

and sometimes it print EMG Data 3 times first and IMU once

So i think the program is collecting the EMG Data in the first sampling time, print it in Serial Monitor, and then collecting the IMU Data in the different sampling time, and print it in Serial Monitor. 
Please correct me if i have something wrong either in my Code or statement. 🙏

vroland, 2019-03-20:
Looks like it's working...? I wrote this a few years ago, so I'm not sure about sampling timing any more. The easiest thing is probably to look at the source code yourself ;)

simbolonmartin, 2019-03-26:
I see..
Thank you
Your library documentation is helping me a lot
👍